ARMmbed/mbed-os-5-docs #32: selecting mbed Studio from the Import Program drop-down on our code samples

loverdeg-ep, 2016-12-15:
https://github.com/ARMmbed/Handbook/blob/5.3/docs/dev_tools/options.md#mbed-studio

"mbed Studio is an mbed Enabled IDE based on Cloud9 and offers a rich development environment. It is currently in alpha and should be considered experimental. You can try it by selecting mbed Studio from the Import Program drop-down on our code samples."

I still hunt for this magical drop-down option every couple days.

loverdeg-ep, 2016-12-22:
Heh!
I finally found it.
[http://studio.mbed.com/](http://studio.mbed.com/)
Should have been obvious I suppose.

iriark01, 2017-01-03:
Thank you for highlighting this. The docs site doesn't integrate with mbed Studio at the moment; I've removed the confusing text.

closed 2017-01-03 by iriark01