ARMmbed/mbed-os-5-docs #92: mbed TLS Porting Guide - link broken

MarceloSalazar, 2017-02-02:
Link at the bottom of this doc is broken:
https://docs.mbed.com/docs/mbed-os-handbook/en/5.2/advanced/tls_porting/

It's pointing at this location (https://docs.mbed.com/docs/mbed-os-handbook/en/5.2/advanced/mbed_OS/Config_sys.md) which is not available.

"
You can read more about how to add a macro for your application ***here***.
"

iriark01, 2017-02-13:
@AnotherButler can you check if it's supposed to point to https://docs.mbed.com/docs/mbed-os-handbook/en/5.2/advanced/config_system/

AnotherButler, 2017-02-13:
@iriark01 That is correct. I have submitted PR #99 to fix this.

AnotherButler, 2017-02-15:
@MarceloSalazar This should now be fixed. Can we close this issue?

MarceloSalazar, 2017-02-16:
Yes, it's fixed now. Thanks!

closed 2017-02-16 by MarceloSalazar