ARMmbed/mbed-os-5-docs #112: Link to mbed_targets.md broken?

ilg-ul, 2017-02-23:
In `docs/advanced/exporters.md` there is 

```
## Eclipse and Make 

Eclipse project export uses a generated Makefile for building. To provide support for 
Make export, you only have to provide the target configuration within the 
[config system](https://github.com/ARMmbed/mbed-os/blob/master/docs/mbed_targets.md).
```

As far as I can tell, the link is broken.

@theotherjimmy, can you clarify what does this phrase mean? I would like to add here a reference to the GNU ARM Eclipse exporter, but I don't know how to cope with this phrase.

theotherjimmy, 2017-02-23:
The phase is supposed to say that support for GCC_ARM implies support for Make_GCC_ARM. I think the same is true for your GNU ARM Eclipse exporter. @AnotherButler Could you help us with phrasing this? That link should be https://github.com/ARMmbed/Handbook/blob/5.3/docs/advanced/mbed_targets.md

ilg-ul, 2017-02-23:
in #113 I suggested to add after that line:

```
## GNU ARM Eclipse managed projects
 
 The GNU ARM Eclipse exporter is available for all targets that use the GCC_ARM toolchain.
```

If you have a better phrasing for both paragraphs, please let me know.

AnotherButler, 2017-02-23:
I fixed the broken link with PR #114. We use relative links when possible. 

@ilg-ul, I like that added wording. 
What do you and @theotherjimmy think about changing "To provide support for Make export, you only have to provide the target configuration within the config system." to "If you have support for GCC_ARM, you have support for Make export." or "If you have support for GCC_ARM, you have support for MAKE_GCC_ARM."?

theotherjimmy, 2017-02-23:
"If you have support for the GCC_ARM toolchain, you have support for MAKE_GCC_ARM." I think it can get confusing if we don't highlight the difference between the exporter and the toolchain.

ilg-ul, 2017-02-23:
the previous thumb up means I agree that it can get confusing.

what is the message you want to transmit? being a page about exporters, I guess you want to say something about the exporter. what exactly? the relationship with toolchains? the fact that a given exporter works with certain toolchains?

theotherjimmy, 2017-02-23:
I'm trying to say that if `mbed compile -t <toolchain>` works then `mbed export -i make_<toolchain>` will also work.

theotherjimmy, 2017-02-23:
It gets worse because `mbed export -i gcc_arm` == `mbed export -i make_gcc_arm`.

ilg-ul, 2017-02-23:
that's a good example why manuals must not be written by programmers ;-)

theotherjimmy, 2017-02-23:
Indeed, just verified by programmers

iriark01, 2017-03-14:
The PR for this is merged - can I close this as resolved?

closed 2017-03-14 by ilg-ul