ARMmbed/mbed-os-5-docs #126: Missing Visual Studio Code Exporter from CLI

pfried, 2017-03-09:
There is an article in the Handbook describing the use of a VSC debugger. 

https://docs.mbed.com/docs/mbed-os-handbook/en/latest/debugging/vscode/

The CLI command

`$ mbed export -i vscode_gcc_arm -m K64F`

should export a program to VSC but there is no such exporter

iriark01, 2017-03-14:
@janjongboom can you help?

janjongboom, 2017-03-17:
@pfried @iriark01 We're waiting on some work by Microsoft... See https://github.com/ARMmbed/mbed-os/pull/3915.

To get it to work right now:

```
cd mbed-os
git remote add janjongboom git@github.com:janjongboom/mbed-os.git
git fetch janjongboom
git checkout janjongboom/feature-vscode
cd ..
mbed export -i vscode_gcc_arm -m K64F
```

After that you can switch back to whatever mbed OS version you were on before.

AnotherButler, 2017-05-08:
@janjongboom @pfried Is this OK to close because ARMmbed/mbed-os#3915 has been merged, or is this still an open issue?

janjongboom, 2017-05-08:
@AnotherButler Yes, can be closed.

BUT... The vscode page is removed from handbook, so it should be reinstantiated.

closed 2017-05-08 by AnotherButler