ARMmbed/mbed-os-5-docs #132: Online compiler compile menu options are described incorrectly

adamgreen, 2017-03-25:
At https://github.com/ARMmbed/Handbook/blob/5.4/docs/dev_tools/online_comp.md#compile-and-download , I see the first two compile options described like this:

1. **Compile:** builds your code and downloads a binary file. Current program only.
1. **Compile All:** same as *compile*, but for all programs.

The difference between these two options isn't what programs in your workspace are built but whether an incremental or clean build is performed. The following descriptions would be closer to what I expect to see here in the documentation:
1. **Compile:** builds code that you have modified since your last compile and downloads the resulting binary file.
1. **Compile All:** same as *compile*, but rebuilds all source code, even if it hasn't changed since the last compile.

theotherjimmy, 2017-03-27:
The latter bullets are correct.

theotherjimmy, 2017-03-27:
Feel free to submit a PR to change this. @AnotherButler, could you comment on which branch that PR would be made against?

AnotherButler, 2017-03-27:
Please make such a PR against the 5.4 branch.

adamgreen, 2017-03-27:
PR submitted against 5.4 branch.

AnotherButler, 2017-03-28:
@adamgreen Thanks. I've merged your PR. Is it OK to close this issue?

adamgreen, 2017-03-28:
Thanks for merging the PR. I will close this bug.

closed 2017-03-28 by adamgreen