ARMmbed/mbed-os-5-docs #133: Permissive Binary License sometimes referred to as Public Binary License

adamgreen, 2017-03-25:
In the https://github.com/ARMmbed/Handbook/blob/5.4/docs/cont/contributing.md#how-to-apply-pbl-correctly section of the Handbook the PBL is sometimes expanded to **Public Binary License** but in most places (including the legal license description), it is referred to as the **Permissive Binary License**.

Examples of where it appears to be named incorrectly in this section:
<pre>Unless specifically indicated otherwise in a file, files are licensed under the Public Binary License, 
as can be found in: LICENSE-permissive-binary-license-1.0.txt</pre>

* The full original [Public Binary License 1.0 text](https://www.mbed.com/licenses/PBL-1.0) in *LICENSE-permissive-binary-license-1.0.txt*.

AnotherButler, 2017-03-27:
@jlovejoy Could you please help me fix the wording?

jlovejoy, 2017-03-28:
Good catch! yes, it should be "permissive" not "public" (or unofficially, PB might refer to someone's initials... ;)
I just submitted a PR, as the header notice needed a bit updating that @pjbakker and I came up with some time ago, but forgot to update it here as well.

AnotherButler, 2017-03-28:
Thank you so much. You are awesome.

AnotherButler, 2017-03-31:
Thanks for raising this issue, @adamgreen. We just merged PR #137, which should resolve this issue.

adamgreen, 2017-03-31:
Thanks @AnotherButler and @jlovejoy.

closed 2017-03-31 by adamgreen