ARMmbed/mbed-os-5-docs #138: Confused description in the chapter of porting

tz-arm, 2017-03-31:
There are some reference commend in the porting chapter as below.
Actually there is no "mbed-os-example-new-target" repository and mbed import will be fail.
Is it a typo which want to use "mbed new" or there was such a repository?

ps:
- mbed import mbed-os-example-new-target
- cd mbed-os-example-new-target\mbed-os
- git checkout master
- git pull
- git checkout -b my-new-target
- git remote set-url origin https://github.com/USERNAME/mbed-os
- cd ..

AnotherButler, 2017-04-21:
@tz-arm I think PR #143 has resolved this issue. Is it OK to close, or is anything still unresolved?

AnotherButler, 2017-04-28:
@tz-arm I am closing this issue. Please don't hesitate to reopen it if anything is still unresolved.

PR #143 (2017-04-21): Wrestled with the porting guide
May or may not have come out ahead

Might resolve #138
closed 2017-04-28 by AnotherButler