ARMmbed/mbed-os-5-docs #153: Source for https://docs.mbed.com/docs/mbed-os-handbook/en/latest/dev_tools/cli/

bmenchaca, 2017-05-01:
I cannot find the source against which to file issues for the Handbook, Dev Tools, CLI.  The closest I could find the documentation is here:
https://github.com/ARMmbed/Handbook/blob/5.4/docs/dev_tools/cli.md

Which refers to:
https://raw.githubusercontent.com/ARMmbed/mbed-cli/master/README.md%7D

Which says 404.

AnotherButler, 2017-05-02:
This page is transcluded from this README: [https://github.com/ARMmbed/mbed-cli](https://github.com/ARMmbed/mbed-cli). When we restructure these docs and move them to the new engine, we should be able to avoid this type of confusing transclusion. In the meantime, please make any PRs for this page against the linked README.

iriark01, 2017-06-05:
Can we close this?

AnotherButler, 2017-06-06:
@bmenchaca Please feel free to reopen this if you feel it has not been resolved fully.

closed 2017-06-06 by AnotherButler