ARMmbed/mbed-os-5-docs #184: I2C Master clarification

noutram, 2017-06-27:
It is clear that I2C Slave is not interrupt safe. What is maybe ambiguous is I2C Master. This document does not state it explicitly, and students have been asking.

https://github.com/ARMmbed/Handbook/blob/5.5/docs/concepts/thread_safety.md#drivers

AnotherButler, 2017-06-27:
@noutram Thanks for letting us know. We'll update our documentation to clarify. In the meantime, [the Doxygen](https://github.com/ARMmbed/mbed-os/blob/master/drivers/I2C.h#L39) shows that I2C Master is Thread safe. Does that help?

noutram, 2017-06-27:
Hi. Thanks for the quick response. Did you mean interrupt safe instead of thread?

0xc0170, 2017-06-28:
@noutram 

Based on the referenced page, there is a list of drivers that are interrupt safe, the rest are either only thread safe or not protected at all (the list of unprotected is there as well). Thus I2C Master should be thread safe, but not interrupt safe (protected by mutex). Does that answer your question? How would you change the document to be more clear?

noutram, 2017-06-30:
I've re-read the documentation and tried to understand why it was misunderstood by both myself and others. It boils down to the fact that people don't always read documents linearly from top to bottom, different definitions of terminology and the titles.

I should stress that the document is very precise.

1. Section titles

The document title of the section is "Thread Safety in mbed OS 5" where for most people, this is about "Thread and Interrupt Safety in mbed OS 5". When a student goes looking for help on interrupt safety, they may skip this this section. Maybe consider changing the title?

You then have a sub-heading "Thread safety". Again, for those less familiar with the precise terminology, consider maybe "Thread and Interrupt safety".

2. Terminology

The subsection "Drivers" is perfectly accurate, but assumes the reader has read the precise definitions in the section "Synchronization levels".  In fact, a Venn Diagram might be helpful (see below)

These definitions are critical, but not everyone is aware that interrupt safety is a superset of thread safety for example. As you read this section, on many devices, the definitions are off screen. It's probably a reality that students will jump straight to the section of interest (Drivers) and miss this. 

Now, one could argue people should read the documentation, but the reality is simply not that (from a few years of teaching). There is no harm in adding some repetition, _especially as this relates to a safety critical topic_.

So, maybe repeat or include the aforementioned Venn Diagram, placing it adjacent to the Drivers section. This means all key information is in the field of view and there is less opportunity to be misunderstood (far more dangerous than not being understood :)

Great work - we all appreciate what you do here.

Would you like me to try and propose some edits and generate a PR?

0xc0170, 2017-06-30:
> Would you like me to try and propose some edits and generate a PR?

Yes, always welcomed !

noutram, 2017-06-30:
Ok, I'll get on to it as soon as I can. I'll try and create a figure as proposed.

noutram, 2017-06-30:
Ok,I've generated a PR on branch 5.5. I also added a figures folder with a new figure + XML source (for draw.io).

AnotherButler, 2017-07-10:
We have merged PR #189 Thanks for creating it. 👍 

Is this issue OK to close?

noutram, 2017-07-10:
Sure. Very happy to help

closed 2017-07-10 by noutram