ARMmbed/mbed-os-5-docs #192: Handbook website and sources mismatch

bulislaw, 2017-07-07:
Content in sources and rendered website differ, it's not due to recent chances to docs, as the docs are quite old.
https://github.com/ARMmbed/Handbook/blob/5.5/docs/dev_tools/build_profiles.md
https://docs.mbed.com/docs/mbed-os-handbook/en/latest/dev_tools/build_profiles/

This part is missing from the website:
```
  * Debugger is likely to drop connection.
  * Breaks the local file system on the [mbed interface](https://docs.mbed.com/docs/mbed-os-handbook/en/latest/getting_started/mbed_interface/) on some boards.
```
for both develop and release profiles.

@AnotherButler

AnotherButler, 2017-07-07:
@bulislaw Great eye. I have updated this. Is it OK to close?

bulislaw, 2017-07-10:
Brilliant, thanks. Closing.

closed 2017-07-10 by bulislaw