ARMmbed/mbed-os-5-docs #202: Documentation in pdf format

maxpeng, 2017-08-08:
I find mbed OS 5 handbook is a great resource to learn mbed OS 5. However it is inconvenient to resume the reading on browser, is there a way to convert the documentation into a pdf file for offline reading?

AnotherButler, 2017-08-08:
@maxpeng I'm so glad you find the handbook helpful. At this time, we do not offer a way to convert the documentation into a PDF. I'm sorry for the inconvenience.

BlackstoneEngineering, 2017-08-10:
@maxpeng Great idea! We have this feature on our backlog of things todo, I will bring this up with our web team to implement in the future. Thanks for your suggestion!

maxpeng, 2017-08-10:
@BlackstoneEngineering I am glad to hear this news!!!

Now I am reading [mbed OS API References](https://docs.mbed.com/docs/mbed-os-api-reference/en/latest/), it is pretty interesting to learn mbed OS 5. On my opinion, it changes the way of firmware development.

AnotherButler, 2017-09-18:
@maxpeng I'm going to close this issue because it is in our backlog. Please feel free to reopen it if you feel there is more to be addressed.

closed 2017-09-18 by AnotherButler