ARMmbed/mbed-os-5-docs #275: Enabling Runtime Stats

loverdeg-ep, 2017-09-28:
From:
[https://github.com/ARMmbed/Handbook/blob/new_engine/docs/tutorials/optimize/memory/runtime_stats.md#L11](https://github.com/ARMmbed/Handbook/blob/new_engine/docs/tutorials/optimize/memory/runtime_stats.md#L11)

```
To enable heap stats:
1. Add the command-line flag -DMBED_HEAP_STATS_ENABLED=1.
2. Use the function mbed_stats_heap_get() to take a snapshot of heap stats.
```

Should this be enabled via command line flag or by adding it to macros in the app config json file?

Also why isn't this defined and described as a macro in [https://github.com/ARMmbed/mbed-os/blob/master/platform/mbed_lib.json](https://github.com/ARMmbed/mbed-os/blob/master/platform/mbed_lib.json)

AnotherButler, 2017-09-28:
@deepikabhavnani @theotherjimmy Could one of you please help clarify?

deepikabhavnani, 2017-09-28:
@loverdeg-ep : As stated in handbook it is a "command line flag". All macros in app config json file will have prefix "MBED_CONF_"

40Grit, 2017-09-29:
@deepikabhavnani why does the recommendation differ for the MBED_MEM_TRACING_ENABLED macro?

Just trying to understand how the mbed cli config system is applied to the OS sources.

deepikabhavnani, 2017-09-29:
@40Grit - MBED_MEM_TRACING_ENABLED is for runtime tracing and is supported for all toolchains and is not target specific, it is more of an application configuration.  -D defines are usually for target/toolchain specific code. 
Heap statistics for IAR were not supported earlier.  However, now if you propose it to be application config and submit issue(enhancement) on github it may be considered. Also you are welcome to submit pull request with changes.

loverdeg-ep, 2017-09-29:
Will make an issue in mbed-os and watch how it's done to get a better understanding.

closed 2017-09-29 by loverdeg-ep