ARMmbed/mbed-os-5-docs #282: "Introduction to Callbacks" Confusion

loverdeg-ep, 2017-10-02:
On the [Platform Overview](https://os.mbed.com/docs/v5.6/reference/platform-overview.html) page in the Reference manual there is a "Callbacks" section that seems out of place.

The [Callback](https://os.mbed.com/docs/v5.6/reference/callback.html) page references the previously mentioned section as the  "the introduction to callbacks".

I found it confusing that Callbacks  were the only platform feature to be introduced as a section in the top level the Reference manual's Platform page.

loverdeg-ep, 2017-10-06:
@AnotherButler 
There are no "linked" subheadings on the [Platform Overview](https://os.mbed.com/docs/v5.6/reference/platform-overview.html) page either.

As an engineer trying to meld with the documentation, little things like this make it really hard to get a flow going.

loverdeg-ep, 2017-10-06:
> As an engineer trying to meld with the documentation, little things like this make it really hard to get a flow going.

"this" being issue #282 as a whole not just the lack of "linked" subheadings.

AnotherButler, 2017-10-06:
@loverdeg-ep Thanks for the great feedback. We have plans to add more content about other platform APIs to the Platform Overview page, so Callbacks will no longer be the only platform feature introduced.

@geky @kegilbert Could one of you please reword the "introduction to callbacks" link to be less confusing?

closed 2017-10-09 by loverdeg-ep