ARMmbed/mbed-os-5-docs #292: Coding guideline: Clarification on pointer position in declarations

pan-, 2017-10-17:
In the coding guideline, under the `rules` section the following rule is enunciated: 

> For pointers, `*` is adjacent to a name (`analogin_t *obj`).

It raise several questions: 

* Should this rule be applied to references ? 
* What if the name is omitted ? This can be the case in function declaration. Shall it be `void foo(int*)` or `void foo(int *)`
* Is that rule valid for function return type ? Shall it be `int *foo(void)` or `int* foo(void)` ? 
* What shall be used to declare immutable pointers ? The rule cannot be applied because the `const` modifier have to go to the right of the pointer according to the C and C++ standards. 
* How pointer to pointer and reference to pointer shall be written `int* &foo` or `int *&foo` ? 

I also want to point out that C and C++ standard takes different stances on the subject (even if both constructs are valid). In the C standard there is always a space between the pointee type and the pointer declarator followed by another space if a qualifiers follow. In C++ standard pointer and references declarator are always weld to the type they point to.

It may be interesting to had a bit of background behind the choice. 

Another note, when doxygen generate a function documentation, the rule is broken and the pointer goes next to the type it points to ([example](https://os.mbed.com/docs/v5.6/mbed-os-api-doxy/classmbed_1_1_serial_base.html#abee1597dcb5e2dfcdfb42ed9fe871e37))

0xc0170, 2017-10-18:
Good points here, we shall update the coding style to have better coverage.

AnotherButler, 2017-10-19:
Good catch. Once there is agreement on answers to all of those questions, please submit a PR against this file: https://github.com/ARMmbed/Handbook/blob/new_engine/docs/reference/contributing/guidelines/style.md

AnotherButler, 2017-12-04:
@pan- @0xc0170 Have y'all reached an agreement on answers to all of these questions? If so, please submit a PR against these files: 

https://github.com/ARMmbed/Handbook/blob/new_engine/docs/reference/contributing/guidelines/design_guidelines.md

https://github.com/ARMmbed/Handbook/blob/new_engine/docs/reference/contributing/guidelines/style.md

AnotherButler, 2018-01-05:
@pan- @0xc0170 Where do we stand on progress made against this issue?

AnotherButler, 2018-01-24:
@pan- @0xc0170 Because I have seen no progress or conversation on this issue in more than a month, I'm closing this issue. Please feel free to reopen if this problem persists.

closed 2018-01-24 by AnotherButler

0xc0170, 2018-01-25:
I proposed a fix to address this issue

PR #382 (2018-02-05): Add pointer and references examples
@pan- This should fix #292 (we might want to add more, let me know). Your issue had it nicely described ,thanks for all that info there.

Reason: the current codebase (most of it) follows what is described here
commit 4f219d25 (2018-02-05): Add pointer and references examples

Fixes #292

Reason: the most of the current codebase follows what is described here.
This should fix few questions we have regarding the alignment of pointers or
references.