ARMmbed/mbed-os-5-docs #331: Need to elaborate on export-able toolchains?

andrewc-arm, 2017-11-09:
Hi,

https://os.mbed.com/docs/v5.6/tutorials/eclipse.html
{quote}
This document explains how to build and debug Arm Mbed OS applications using Eclipse. Before starting, please configure your local debug toolchain.
{quote}

The above text should be elaborated like following.
{quote}
This document explains how to build and debug Arm Mbed OS applications using Eclipse. Before starting, please check which debug toolchains you are targeting from this list.
{quote}

Or maybe I am not understanding the intention well. Please elaborate.

theotherjimmy, 2017-11-09:
Also, It looks like you tried to use asciidoc in your comment, when github only supports markdown.

andrewc-arm, 2017-11-10:
OK. Thanks. That part of the text made me to read exportation page for long which was not needed. Please improve.

AnotherButler, 2018-01-08:
I'm closing this because PR #373 resolves this. Please feel free to reopen this issue if you feel it is not resolved.

closed 2018-01-08 by AnotherButler