ARMmbed/mbed-os-5-docs #335: events_tutorial.md does printf() in ISR, this does not work

janjongboom, 2017-11-13:
In https://github.com/ARMmbed/Handbook/blob/5.6/docs/tutorials/using_apis/events_tutorial.md we do `printf()` calls in ISRs to show how the code is ran. This is not supported. It used to work in GCC but not with latest mbed OS (throws 'not supported in interrupt context' or something similar on the serial) and never worked in ARMCC.

I'm not sure who wrote this article, but it needs to be rewritten.

@AnotherButler

AnotherButler, 2017-11-13:
@janjongboom I wrote this based on a combination of old forum issues, blog posts and comments. It's not a new page, and the content I used to create it was even older, so I'm not surprised it's out of date now. Please feel free to update it to make it technically correct or to let me know exactly how to change it to make it correct. Thanks for catching it 👍

AnotherButler, 2017-12-04:
@janjongboom There is a PR up to improve this document: https://github.com/ARMmbed/Handbook/pull/349

Does it solve this issue?

janjongboom, 2017-12-06:
@AnotherButler, no that's about other documents (same subject, different document). Maybe we should change the current tutorial to https://os.mbed.com/blog/entry/Simplify-your-code-with-mbed-events/, it does not suffer from the same problem.

AnotherButler, 2018-01-05:
@janjongboom Because there is additional content on that page that is still correct, I don't want to replace the whole page. I understand how busy you are, but when you get the chance, could you please help me update the examples and other text in the tutorial that's incorrect? Or could you recommend an engineer who could? Thanks so much.

janjongboom, 2018-01-05:
@AnotherButler I'll try and do it next week.

0xc0170, 2018-01-25:
> @janjongboom Because there is additional content on that page that is still correct, I don't want to replace the whole page. I understand how busy you are, but when you get the chance, could you [...]

Update example referenced above, please review

coisme, 2018-01-30:
Since printf() was removed from rise_handler(), the output at the bottom of the source code is inconsistent. It shouldn't show the line "rise_handler in context 0x0".

0xc0170, 2018-01-30:
@coisme Good point. I believe what we can do is fix the example by using RawSerial and print what was previously there to see the difference between contexts.

0xc0170, 2018-05-03:
I believe it should and  this shall be closed. reopened if there is still a problem

PR #384 (2018-01-26): events: latest event example fixes IRQ printf issue
Fixes #335
commit 68aa2835 (2018-01-26): events: latest event example fixes IRQ printf issue

Fixes #335
closed 2018-05-03 by AnotherButler