ARMmbed/mbed-os-5-docs #359: Dis-intuitive handbook navigation

loverdeg-ep, 2017-12-12:
I don't have time to study and explain why I believe this isn't working.
I just know it must not be working because it completely inhibits me from getting into [flow](https://en.wikipedia.org/wiki/Flow_(psychology))

I was checking out the recent content improvements to the handbook and had to literally close the tab to prevent myself from trying to figure out what I was looking at.

While I suspect there might be an issue with the heading levels in this instance, I think my main concern has something to do with the blue lines as a whole.

[image]

AnotherButler, 2017-12-12:
@loverdeg-ep Thank you for your feedback. There is an issue with the heading levels here, and we are currently working on a fix. I'll pass your comment about the blue lines on to our UX team. I understand that you're busy right now. In the future, if you have further feedback about this issue, we're happy to hear it.

loverdeg-ep, 2017-12-14:
@AnotherButler Another note, I've always been a massive fan of the [android developer documentation](https://developer.android.com/index.html).

In my opinion that should be your "gold standard".

You've already developed an engine for documentation.
But if you want the community involved it would be nice if either a popular platform were chosen or 
your documentation standards and procedures were published.

I'll be happy to close when the proper issue or fix is posted regarding the heading level confusion.

loverdeg-ep, 2017-12-21:
@AnotherButler 

[image]

I just noticed the the [docker docs] (https://docs.docker.com/engine/breaking_changes/) use a similar navigation scheme to yours.

I think the drop-downs make all the difference and I like the bold line indicating the current page versus mbed docs 'light gray highlight'.

Sorry; not sure why but this is a sensitive subject for me.

AnotherButler, 2018-01-05:
@loverdeg-ep I am happy to report that our team has fixed the problem with the navigation bar headings. You can find the correctly formatted page here: https://os.mbed.com/docs/v5.7/reference/apis.html

I have passed your feedback on to our UX team. Thank you for taking the time to create this issue. We appreciate your input. If you have any other suggestions, please don't hesitate to create another issue or pull request.

closed 2018-01-05 by AnotherButler