ARMmbed/mbed-os-5-docs #362: BLE class reference gives page error in v5.6 docs

senthilr, 2017-12-15:
Clicking on the BLE Class Reference url throws "Page error" and the class reference does not render.
https://os.mbed.com/docs/v5.6/mbed-os-api-doxy/class_b_l_e.html
This seems to be the case for all BLE APIs
https://os.mbed.com/docs/v5.6/mbed-os-api-doxy/class_heart_rate_service.html
https://os.mbed.com/docs/v5.6/reference/ibeacon.html#ibeacon-class-reference
@AnotherButler

AnotherButler, 2017-12-20:
@senthilr Thanks for raising this issue. Because of an mbed-os policy to exclude Doxygen that throws warnings, these APIs were not included in the 5.6 Doxygen. However, we have fixed this issue in 5.7: https://os.mbed.com/docs/v5.7/reference/ble.html

AnotherButler, 2018-01-02:
Please don't hesitate to reopen this issue if you feel it is not resolved.

closed 2018-01-02 by AnotherButler