ARMmbed/mbed-os-5-docs #365: Sleep documentation could be improved - and is not findable

janjongboom, 2017-12-18:
Sleeping in Mbed OS 5 is fundamentally different from earlier versions (SleepManager.md). We need to emphasize that `sleep()` should never be called manually and add some extra examples how this works with multiple threads pinging. Also, how does sleep behave when you wait on a Semaphore instead of via wait()? Also, need more info that on `develop` profile the device might not go to sleep properly, and how to enable release mode from online compiler.

Another issue is that the new docs are unfindable through Google: https://www.google.co.uk/search?q=sleep+mbed&oq=sleep+mbed&aqs=chrome..69i57.1087j0j1&sourceid=chrome&ie=UTF-8 We need to put some notes on the old pages that this has changed now.

@AnotherButler @senthilr

AnotherButler, 2017-12-18:
@janjongboom Thank you for the issue. I'll create a ticket for this. In the future, please create a ticket for similar issues.

"We need to put some notes on the old pages that this has changed now."
Do you think we need more than the big yellow bar with the exclamation mark and note to indicate that the documentation has moved? 

[image]

AnotherButler, 2017-12-18:
@janjongboom There is an open PR that I think may answer the questions you raised. Does this solve this issue?
https://github.com/ARMmbed/Handbook/pull/351

janjongboom, 2017-12-18:
@AnotherButler, well that's the build profiles page, I wouldn't expect sleep info to be there... But f.e. the SiLabs sleep page could use a mark.

janjongboom, 2017-12-18:
@AnotherButler #351 is a start, but it still lacks information and code samples beyond the most basic of use cases.

AnotherButler, 2017-12-18:
@janjongboom Thanks for clarifying. I'll tag you in the ticket I create. The technical authors do not control the content of the teams pages. I'll talk to Partner Enablement about those.

closed 2018-01-11 by AnotherButler