ARMmbed/mbed-os-5-docs #375: Serial - example three - printf in ISR

0xc0170, 2018-01-09:
Reference: https://github.com/ARMmbed/Handbook/blob/5.7/docs/reference/api/drivers/Serial.md#example-three

It should either be doing buffering in ISR or use RawSerial (the api is regarding Serial thus would probably need to be rewritten to capture characters and process them in the main thread?)

AnotherButler, 2018-01-09:
If this is incorrect, please update this example or let me know who to tag to do it.

0xc0170, 2018-01-10:
Done, Pr created : https://os.mbed.com/teams/mbed_example/code/Serial_ex_3/pull-request/1

@sarahmarshy can you review (I noticed your last commit there)

sarahmarshy, 2018-01-10:
@0xc0170, my last commit just updated Mbed OS. Your changes Lgtm

0xc0170, 2018-01-15:
The example was updated, we just need now to point docs to this revision: https://os.mbed.com/teams/mbed_example/code/Serial_ex_3/rev/4ab47f33a1ae/

0xc0170, 2018-01-25:
Proposed fix referenced above, please review

AnotherButler, 2018-01-25:
@0xc0170 Thank you for PRs. I have merged them and made the changes live, so I am closing this issue. Please feel free to reopen it if it is not resolved to your satisfaction.

PR #383 (2018-01-25): Serial: fix serial 3 example - use rev2
Fixes #375
closed 2018-01-25 by AnotherButler