ARMmbed/mbed-os-5-docs #379: Navigation Panel - Current Page Indicator too light

loverdeg-ep, 2018-01-24:
In the image below, I couldn't see the current page indication "light gray-blue highlight".

Looking at this on a different monitor indicated my monitor needed adjusting.

However, I still think the current page indication is either too light or needs a different indicator altogether.

[image]

AnotherButler, 2018-01-24:
@loverdeg-ep Thank you for your feedback. We appreciate your taking the time to let us know about this. I am passing this information on to our UX team.

AnotherButler, 2018-01-24:
@loverdeg-ep I have spoken with our UX team. They say there are planned usability improvements that are related to our table of contents navigation, and they have added this as something to look into. Thank you again for raising this issue.

closed 2018-01-24 by AnotherButler