ARMmbed/mbed-os-5-docs #380: Inconsistent Page Leveling

loverdeg-ep, 2018-01-24:
The following screen capture comes from the page shown when I select [Reference](https://os.mbed.com/docs/v5.7/reference/index.html) in the navigation pane.

[image]

As shown selecting [Reference](https://os.mbed.com/docs/v5.7/reference/index.html) actually takes you down 2 extra levels to the "MBED OS Reference Book" page.

All other top level Navigation links (Introduction, Tools, Tutorials) behave similarly.

Compare this behavior to when [Configuration](https://os.mbed.com/docs/v5.7/reference/configuration.html) is selected.

[image]

I would greatly prefer the behavior exhibited in the latter example for the top level navigation links.

Let me know if this is confusing.

loverdeg-ep, 2018-01-29:
@AnotherButler Does this follow under the "planned usability improvements that are related to our table of contents navigation" from #379?

If so please close.

AnotherButler, 2018-01-29:
@loverdeg-ep Yes, this does. I think there may be a temporary workaround I can do from my end in the meantime. I'm going to leave this ticket open while I explore this option, but if it doesn't work, we'll have to wait for the planned improvements.

AnotherButler, 2018-04-05:
Closing until planned usability improvements are complete

closed 2018-04-05 by AnotherButler

AnotherButler, 2018-04-11:
@loverdeg-ep Our UX team says we can expect to see changes in June.