ARMmbed/mbed-os-5-docs #395: documentation for mesh is broken

hasnainvirk, 2018-02-03:
Main issue, documentation for mesh is broken. Breakdown of APIs is inconsistent.
https://os.mbed.com/docs/v5.7/reference/network-socket.html NetworkInterface does not list mesh.

https://os.mbed.com/docs/v5.7/reference/network-socket.html, In network socket list, link for the mesh is broken. 

As a general comment, I would like to point out that the current style of documentation for network technologies is unattractive and 'way too' concise. Documentation for other bits of operating system is fine but network technologies which are the heart of all  what IoT stands for is dismal.

AnotherButler, 2018-02-04:
Because of the errors and warnings it throws, the mesh API has been excluded from the Doxygen. Because we can't link to something that isn't there, there is no link, as you saw. The core OS team controls the Doxygen exclusion. If you'd like to create a PR to fix this issue, I'm sure they'd be glad to merge it as soon as possible.

Also, much of the second half of your comment seems unnecessary. Please stick to the facts. If you'd like to create a PR to improve our documentation, I'd be happy to review it. Please note that we also have mesh documentation in our tutorials section.

hasnainvirk, 2018-02-07:
How is that unnecessary ? I'm giving you an honest developer feedback. Which facts should I stick to ? 
My intention is to pass constructive criticism. Take it or leave it.

kjbracey-arm, 2018-02-07:
IP core team (and others) are planning a work package for 5.9 to do general improvements on the networking-related stuff in the handbook.

AnotherButler, 2018-02-07:
@kjbracey-arm Because this work is planned, is it OK to close this issue?

@hasnainvirk We appreciate honest feedback but require objective statements to make changes. "Unattractive" and "dismal" are subjective. What is unattractive to one person may be attractive to another. If you can explain in objective terms what you would like fixed or, even better, put up a PR to show us how you think things should look, we're happy to review your feedback and suggestions.

AnotherButler, 2018-03-20:
Now that the code and Doxygen have been fixed, you can find the mesh API at https://os-doc-builder.test.mbed.com/docs/development/reference/mesh-api.html

closed 2018-03-20 by AnotherButler