ARMmbed/mbed-os-5-docs #405: Make this repository searchable

pan-, 2018-02-09:
Github search engine [restrict](https://help.github.com/articles/searching-code/#considerations-for-code-search) indexation for search to a repository default branch. 

This repository default branch, `master`, is empty and is not updated. All developments happen in the `new_engine` branch. 

Would it be possible to make the `new_engine` branch the default branch or bring back development on master so we can search the repository via Github ?

loverdeg-ep, 2018-02-09:
Agree.
Also, is new_engine named for the new documentation generator engine?
If you are already on the new_engine would it make sense to use a different default branch?

AnotherButler, 2018-02-14:
@iriark01 Could you please help me look into making the `new_engine` branch the default branch?

iriark01, 2018-02-15:
I made it default.

closed 2018-02-15 by iriark01