ARMmbed/mbed-os-5-docs #406: Documentation standards and procedures repository

loverdeg-ep, 2018-02-09:
@AnotherButler There used to be a public repository for meta-documentation of this repository.

Can it be made public again?

I believe it also documented how the "new_engine" operates which would be useful for the community to make better merge requests.

AnotherButler, 2018-02-14:
@loverdeg-ep The only public documentation I can think of is our [contributing guidelines](https://os.mbed.com/docs/v5.7/reference/guidelines.html).

@iriark01 Can you think of any other public documentation that we have about contributing to our documentation?

iriark01, 2018-02-15:
There used to be, back when we had the MkDocs engine. But we've removed it.

AnotherButler, 2018-02-15:
@loverdeg-ep We do not have any public meta-documentation at this time. Do you think it would be beneficial to add information about contributing to our documentation in addition to the information about contributing to our code base?

iriark01, 2018-02-15:
There is a link to GitHub on the docs page but I guess it has no info - it's just a link.

loverdeg-ep, 2018-02-16:
@AnotherButler @iriark01 
I'm pretty sure the docs in the repository I was referring to specifically mentioned the new engine.
They also described the tone that should be used, words to avoid, and workflow for documentation releases.

It was only public for about a week.
I'll try and dig up the link form my history.

iriark01, 2018-02-16:
That isn't public because it contains internal info.

loverdeg-ep, 2018-02-16:
Knew I should have forked or pulled while I had the chance...

@AnotherButler  Again we won't really feel comfortable making pull requests unless we understand how the engine works and what your documentation standards are. 

Feel free to close.

iriark01, 2018-02-16:
You don't have to worry about the engine - it's what we're here for.

closed 2018-02-16 by AnotherButler