ARMmbed/mbed-os-5-docs #409: Documentation for using vsCode exported project with GCC ARM Embedded Toolcahin on Windows

loverdeg-ep, 2018-02-13:
@janjongboom 
I have a customer who would prefer to use windows to host their toolchain.

The documentation for vscode debugging seems to indicate that all you'll need to do is install ARM embedded toolchain, export for vscode and everything will build. Which would probably be the case in linux.

However, I was under the impression that ARM embedded toolchain windows installer doesn't include make.

Can the documentation be clarified as to how to set up the ARM embedded toolchain to work with an exported vscode project on windows?

As of this moment it seems to only link to mcu on eclipse documentation which is eclipse specific.

References:
[https://os.mbed.com/docs/v5.6/tutorials/visual-studio-code.html](https://os.mbed.com/docs/v5.6/tutorials/visual-studio-code.html)
[https://os.mbed.com/docs/v5.6/tools/setting-up-a-local-debug-toolchain.html](https://os.mbed.com/blog/entry/Adding-a-Visual-Studio-Code-exporter/)
[https://os.mbed.com/docs/v5.6/tools/setting-up-a-local-debug-toolchain.html](https://os.mbed.com/docs/v5.6/tools/setting-up-a-local-debug-toolchain.html)
[https://os.mbed.com/docs/v5.6/tools/exporting.html](https://os.mbed.com/docs/v5.6/tools/exporting.html) (IMO especially confusing document, maybe break it into multiple docs for each exporter)

loverdeg-ep, 2018-02-13:
@janjongboom 
Also out of curiosity, why export this as a make file project versus having vscode trigger an mbed-cli build?
intellisense?

janjongboom, 2018-02-14:
@loverdeg-ep We use make because exported projects don't depend on Mbed CLI. You can however easily build using CLI, just change the script in launch.json to call `mbed` (with `--profile=debug` as arguments) instead. I use that all the time.

You're correct that we miss instructions to install Make on Windows. Will add it.

janjongboom, 2018-02-14:
FIxed in https://github.com/ARMmbed/Handbook/pull/410

loverdeg-ep, 2018-02-14:
@janjongboom we are already replacing the build command, for both eclipse and vsCode exported projects, with a command to run mbed-cli and toolchain in a docker container.

Our concern was more for keeping the customer comfortable by using familiar tools.

AnotherButler, 2018-02-22:
Closing because PR #410 has merged. Please feel free to reopen if this is not resolved to your satisfaction.

closed 2018-02-22 by AnotherButler