ARMmbed/mbed-os-5-docs #456: nRF Master Control Panel is now called nRF Connect

loverdeg-ep, 2018-03-15:
[bluetooth.md] (https://github.com/ARMmbed/Handbook/blob/new_engine/docs/reference/api/connectivity/bluetooth/bluetooth.md) has a reference to nRF Master Control Panel which is now called nRF connect.

Also, I've seen similar references in mbed BLE examples elsewhere.

AnotherButler, 2018-03-19:
@loverdeg-ep Great catch. 

@pan- Could you please update these? Thanks so much.

AnotherButler, 2018-06-15:
@loverdeg-ep Have these changes addressed this issue?

loverdeg-ep, 2018-06-29:
@AnotherButler I'm no longer seeing any references to "nRF Master Control Panel" in the documentation.

I do see references in many of the BLE example readmes but those are outside the scope of this remote.

closed 2018-06-29 by loverdeg-ep