ARMmbed/mbed-os-5-docs #461: https://os.mbed.com/docs/latest/tutorials/quick-start-offline.html has a broken link for the download

UDXS, 2018-03-19:


AnotherButler, 2018-03-19:
@UDXS Could you please clarify which link is broken? I can open the Windows installer link, the Python link, the CoolTerm link and the PuTTy link. Is one of those giving you trouble, or is it another?

UDXS, 2018-03-20:
It was the Windows Installer one. It gives a 404 error.

AnotherButler, 2018-03-21:
@UDXS Could you please try that link again here: https://os.mbed.com/docs/latest/tutorials/quick-start-offline.html

I'd like to be sure that's fixed. Thanks so much.

UDXS, 2018-03-22:
@AnotherButler Sorry, but it doesn't appear to work.

[Here is a video of it not working](https://youtu.be/J4dqwj2JMPU)

Thanks!

maclobdell, 2018-03-26:
I can confirm the issue.   

A link at the following page is bad: 
https://os.mbed.com/docs/latest/tutorials/quick-start-offline.html

The section is: 

Windows
1. Install Mbed CLI
Download and run the Mbed CLI Windows .exe installer.

The bad link is this going to:
https://github.com/ARMmbed/mbed-cli-windows-installer/releases

The reason it is bad is because it is a private repo.  not sure why it is private, but it looks like there is an alternative section in the docs that had a link to a windows installer that is good.  

See location: 
https://os.mbed.com/docs/v5.7/tools/installing-with-the-windows-installer.html

Section: 

Running the installer
Download the latest executable from mbed-windows-installer v0.4.3.
Run mbed_installer_v042.exe.

Please fix asap!  This is a critical one.

Thanks!

AnotherButler, 2018-03-26:
Does this PR adequately fix the issue? https://github.com/ARMmbed/Handbook/pull/469

UDXS, 2018-03-26:
Yes, I believe it does.
I will close this as long as that PR is accepted.
Thank you, @AnotherButler and @maclobdell

closed 2018-03-26 by UDXS