ARMmbed/mbed-os-5-docs #494: EventQueue examples not ordered correctly

bmcdonnell-ionx, 2018-04-12:
On this page

https://os.mbed.com/docs/v5.8/reference/eventqueue.html

in the example "Shared event example: deferring from interrupt context", it says, "Like the previous example, this defers from interrupt to an event queue thread." However, that's not what either of the two preceding examples do.

I think it should immediately follow the first example, which is titled "EventQueue example: deferring from interrupt context". The titles differ by only 1-2 words, and the code only by a few lines.

bmcdonnell-ionx, 2018-04-12:
Also, it would be helpful if you could provide an example using the `EventQueue::background()` function. 😄

AnotherButler, 2018-04-13:
@bmcdonnell-ionx Thanks for raising this issue. We'll take care of this at once. 

@SenRamakri @kegilbert I can move the example up to remove the confusion in the first comment. Could one of you or someone on your team please create an example using the `EventQueue:background()` function? I can create a ticket if you like.

AnotherButler, 2018-05-01:
@bmcdonnell-ionx Thanks again for raising this issue. 

We have moved the example up to prevent further confusion. We've also created an internal ticket to ensure completion of an example using `EventQueue:background()`.