ARMmbed/mbed-os-5-docs #499: document which classes use dynamic memory

bmcdonnell-ionx, 2018-04-19:
### Request

Document which classes/functions use dynamic memory.

### Motivation

Developers often avoid or minimize use of dynamic memory in embedded system, to avoid memory fragmentation or overflow, to aid in determinism, etc. Also, an easy way to manage external memory is to make all dynamic allocations go there, and to have all static/stack allocations go to internal memory.

Developers should be able to choose consciously whether and when to use dynamic memory. Thus we need to know when the OS is using it.

bmcdonnell-ionx, 2018-04-20:
The thing which prompted this issue report was me thinking that `mbed::Callback` was using dynamic memory, but it seems I was mistaken about that, as it's using [placement `new`](http://en.cppreference.com/w/cpp/language/new#Placement_new).

This would probably still be good to document, though.

AnotherButler, 2018-04-20:
@bmcdonnell-ionx Thanks for raising this request. We'll add it to our list of planned content.

AnotherButler, 2018-04-20:
@SenRamakri Shall I create this task and assign it to you for now, or is there someone well-suited to create this content at this time?

kegilbert, 2018-07-31:
Did a quick run through, seeing the following modules using dynamic memory:

- EventQueue
- Platform
    - MbedStats (Not enabled by default)
- Features Module [Contained by far the majority of malloc calls]
    - Cellular
        - SMS
        - Network
    - LWIPStack
    - BLE
    - Frameworks
        - mbed-coap
    - Filesystems
        - ChanFS
        - FATFS
        - LittleFS
     - Nanostack
     - NetSocket
         - nsapi DNS
         - Emac drivers

Haven't broken this down into class level yet, does this look accurate to you @SenRamakri ? 
Pinging @geky because this came up in conversation during the memory optimization hackaton (in reference to dynamic memory usage) and he has more history on these modules/would know who to ping for more coverage.

geky, 2018-07-31:
It's probably a better idea to start with the classes we know _doesn't_ use dynamic memory. Nearly every class uses dynamic memory in some form, unless it's related to IRQ safety (Callbacks, Events, RTOS classes). Though removing malloc usage would be an interesting thing to prioritize (@SenRamakri?).

Also note, a lot of classes default to dynamic memory, but can use statically allocated buffers if they are provided. This is true for the EventQueue and I think Threads.

cc @c1728p9, @pan-, @kjbracey-arm

AnotherButler, 2018-08-09:
ping @c1728p9 @pan- @kjbracey-arm