ARMmbed/mbed-os-5-docs #544: EMAC Porting section & enhancements

MarceloSalazar, 2018-05-30:
Things to do:

- EMAC tests: this is an important document and should be part of the porting guide docs in the Handbook
https://github.com/ARMmbed/mbed-os/tree/master/TESTS/network/emac

Probably a sub-section under this doc:
https://os-doc-builder.test.mbed.com/docs/development/reference/ethernet-port.html

- Remove reference to Arm Mbed Client example - this will be deprecated soon
Instead, we'll need to add some kind of stress tests, when it becomes avaialble.

- Minor: Remove / merge duplication of `TESTS/network/emac` ?

FYI @SeppoTakalo @AnotherButler

SeppoTakalo, 2018-05-30:
I disagree.
Testcases are not documented in Handbook. EMAC is not an exception.

Bulk of our testcases are not documented at all. Testcases live in Mbed OS repository and are under constant maintenance. So if we would modify test case, we would be forced to change our Handbook.

We should be able to fix and create new test cases without touching documentation. So only requirement can be the instructions on how to run tests.

MarceloSalazar, 2018-05-30:
My suggestion is on the possibility to automatically bring the *content* from Readme.md to the docs in the web.

There is really good information and I'm a bit concerned might be forgotten - I didn't realize about this content until someone shared a specific link with me.

It shouldn't make a difference to the engineers working on this. It's probably a task for the docs/web teams integrating the docs.

MarceloSalazar, 2018-06-08:
Internal reference: IOTDOCS-513