ARMmbed/mbed-os-5-docs #546: on-line compiler question

jflynn129, 2018-05-30:
Sorry if this isn't the correct place to post this, but I couldn't find a better location.
I have a program that I want to be compiled from either the CLI or the on-line compiler (via mbed.com).  But some of the code that is pulled in by the CLI tools (GCC_ARM) isn't available when using the on-line compiler, so I was looking for a define that indicates that the compiler being used is uniquely the "on-line compiler'
Is there such a define/flag that I can test in the code to conditionally include code?  Basically,

#ifdef online_compiler
do this...
#else 
do that...
#endif

theotherjimmy, 2018-05-31:
@jflynn129 We don't have such a define. Note that the online compiler uses Arm Compiler 5.06 u3, and if you're using GCC 6.whatever you're going to pull in different hal/cmsis implementations. These are in directories that are named `TARGET_ARM` for ARMC5 and `TARGET_GCC_ARM` for GCC. If you point me towards a particular project and it's public, I can provide you with help on the particular problem you're running into.

------------

General design note: I would consider 
```C
#ifdef online_compiler
do this...
#else
do that...
#endif
```
an anti-pattern as it duplicates the code across environments that are only superficially different. The online compiler uses the Mbed OS build tools internally, and any differences (other than ARMC5 vs GCC_ARM) are bugs.

jflynn129, 2018-05-31:
I agree with you design note, but couldn't fix it any other way.  I added a macro **ONLINE_COMPILER** to my mbed_app.json file, and put the following in my code:
```
#ifdef ONLINE_COMPILER   //only do this for on-line compiler

struct timeval { 
    long int tv_sec;
    long int tv_usec;
    };
#else  //else ONLINE_COMPILER
 
#include <sys/time.h>
#include <sys/select.h>

#endif  //end ONLINE_COMPILER
```

as these includes are apparently no contained within the Arm Compiler 5.06 toolchain. Perhaps this is a legit difference between ARMC5 and GCC_ARM?

theotherjimmy, 2018-05-31:
@jflynn129 adding the macro to `mbed_app.json` add it both online and offline. So that's not a great way to distinguish.

@bulislaw Do we have a time API for something like this?

jflynn129, 2018-05-31:
yes, I have to remove the macro when working with the mbed cli

bulislaw, 2018-06-01:
Differences between GCC and ARM sound like a bug, there should be no differences from user point of view. @jflynn129 could you share more details about what are you seeing?

jflynn129, 2018-06-01:
when compiling with the ARM compiler:
**Error: Cannot open source input file "sys/time.h"**: No such file or directory in "platform/common/timer_platform.h", Line: 36, Col: 23
but no problem with GCC_ARM using CLI.  Was fixed with the conditional compiler mentioned above. 

Can be reproduced by importing into the on-line compiler from:
https://github.com/Avnet/aws-iot-device-sdk-mbed-c
remove the ONLINE_COMPILER   in the mbed_app.json file and compile.

theotherjimmy, 2018-06-01:
@bulislaw Aside from code size and speed of course.

----------

@jflynn129 Yes the online compiler understands `.mbedignores`. It uses the same code to compile as Mbed CLI.

jflynn129, 2018-06-01:
@theotherjimmy how do you find and edit it with the on-line compiler?  It seems to be hidden and un-editable.

theotherjimmy, 2018-06-01:
That's a really good question. @thegecko ?

thegecko, 2018-06-01:
Correct, you can only create files with known extensions. A *very* hacky way around this is that you can drag and drop files from your desktop into your program. However, as it's an unknown file type, you can't edit it nor replace it, so it would be a one-shot process.

@theotherjimmy Would you mind raising an IOTTOO issue and we'll look into adding .mbedignore support to the online compiler.

iriark01, 2018-06-04:
Hi all,

Can I close this issue now? Looks like @theotherjimmy created a Jira ticket for it.

jflynn129, 2018-06-04:
Ok by me. Would like to know when it will be possible to edit the mbedignore file though.

theotherjimmy, 2018-06-04:
@iriark01 Let's keep this open until @jflynn129 Is satisfied that it has been fixed.

iriark01, 2018-06-04:
Can you maybe move it to the relevant tool's repo? This isn't a good place to keep track of it.

theotherjimmy, 2018-06-04:
We can move it to Mbed OS. That tools repo is not public.