ARMmbed/mbed-os-5-docs #550: Minor issues on Cellular arch doc

MarceloSalazar, 2018-05-31:
Having reviewed the cellular architecture documentation:
https://os-doc-builder.test.mbed.com/docs/development/reference/cellular.html

Find here some feedback:

> Mbed OS cellular is the preferred choice of building applications for Mbed Cloud and cellular IoT solutions

This is not relevant and not true.
Depends on customer requirements for their projects.
We shouldn't mix topics - can we please rephase/remove?

> Out-of-the-box cellular modem support

What does it mean?
Does it work regardless of which operator you have in your area?
Is it expected to work worldwide?
Is it a matter of powering up and connecting automatically?

> Mbed OS already supports several Mbed Enabled boards with on-board cellular modules out of the box

Can we please add a link to those platforms.
For Mbed platforms: https://os.mbed.com/platforms/?connectivity=1
For Mbed modules: https://os.mbed.com/modules/

> If you are using an Mbed OS target that has a supported on-board (mounted) cellular module

This topic is a matter of terminology and alignment with products.
I think we should use cellular 'modem' instead of 'module'.

@MarkJWade as you're looking after modules, can you please comment on this?
You may want to review other parts of this doc as well.

> PPP mode with the LWIP stack on Mbed OS

It would be great to add this image, as not everyone will be looking into the porting guide.
https://s3-us-west-2.amazonaws.com/mbed-os-docs-images/Cell_PPP.png

> AT mode with the IP stack on the modem 

Same, it would be great to include this image, as not everyone will be looking into the porting guide.
https://s3-us-west-2.amazonaws.com/mbed-os-docs-images/Cell_AT.png

> LWIP

This is a very minor issue on naming - there are different variances: LWIP, LwIP, lwIP, ...
@AnotherButler , we should try to align on this - FYI see https://github.com/cvra/lwip

> Sleep more, save energy
> "you may want to optimize it further when implementing applications for constrained battery-operated devices"

This seems a relatively 'complex' handshake.
Can we give a few tips on areas that people should be looking at when optimizing, or just say, for example xxxx
Maybe point to another advanced / porting document that describes this info in a bit more detail?

> Class reference
> mbed::CellularDevice Class Reference

It doesn't make sense to me having the API reference in this section.
It feels unrelated to the audience of this document.
Please consider moving this API somewhere else, for example here:
https://os-doc-builder.test.mbed.com/docs/development/reference/cellular-api.html

MarkJWade, 2018-05-31:
> @MarkJWade as you're looking after modules, can you please comment on this?
You may want to review other parts of this doc as well.

I think the correct term for these cellular modules would be "Hosted Modules", which distinguishes them from the "Hostless Modules" or "Self-Hosted Modules" which the Modules Programme are focusing on.

AriParkkila, 2018-06-11:
Please see fixes in https://github.com/ARMmbed/Handbook/pull/588

AnotherButler, 2018-06-11:
Closing because both PRs merged; please feel free to reopen if you feel this is not resolved

closed 2018-06-11 by AnotherButler