ARMmbed/mbed-os-5-docs #551: Minor issues/questions on Cellular arch doc

MarceloSalazar, 2018-05-31:
Re: Lora Arch documentation:
https://os-doc-builder.test.mbed.com/docs/development/reference/lora-tech.html

> Mbed LoRaWAN solution

The title should be aligned with other names in the same section.
We should remove 'Mbed' and use just 'LoRaWAN'.
Example: 6LoWPAN.

> Arm provides support for SX1272 and SX1276 LoRa radios

Can we add a link:
https://github.com/ARMmbed/mbed-semtech-lora-rf-drivers

>  For a detailed discussion of these events, please visit the LoRaWAN events **documentation**.

Link broken:
https://os-doc-builder.test.mbed.com/docs/development/reference/lorawanevents-api.html

> Receiving messages
> "There are two types of receive() methods in the stack. One is POSIX-like, and you need to tell at what port (instead of a socket ID in Posix format) you wish to receive"

Which one is POSIX-like?
Maybe say the first is POSIX-like.

> Question on LoRaPHY vs Lora drivers

I understand we deliver 10 implementations in Mbed OS of the LoRaPHY class as per specs.
However, how does it relate to the Lora radios that are out of Mbed OS?

I thought the Radio driver was defined at PHY/MAC layer, similar to the 15.4 drivers we have (e.g. [atmel-rf-driver](https://github.com/ARMmbed/atmel-rf-driver))

From section:
" Mbed LoRa radio driver: constructed by the application and passed down to stack."

I think we should expand a bit and explain the difference with PHY layer.

@AnotherButler  @hasnainvirk

AnotherButler, 2018-06-05:
PR up: https://github.com/ARMmbed/Handbook/pull/572

@hasnainvirk Could you please respond to the questions about how LoRa radios outside of Mbed OS relate and about the difference with the PHY layer? If it's easier, feel free to respond in the comments here, and I can add your response to the document.

hasnainvirk, 2018-06-06:
@AnotherButler @MarceloSalazar Sorry for he late reply. Really preoccupied with stuff at hand. LoRaRadio and LoRaPHY having no connection whatsoever in hierarchical terms. LoRaPHY is the network stack layer that controls the LoRaRadio using hooks provided by the LoRaRadio.

MarceloSalazar, 2018-06-08:
@SeppoTakalo as we talked this week, would you be able to help to expand @hasnainvirk 's comments on LoRaPHY vs LoRaRadio and send a PR?
This is useful to people (like me) that don't understand the subtle differences in Lora low-level drivers.
Note, this can happen after 5.9 and it would really help, thanks!

AnttiKauppila, 2018-08-21:
@MarceloSalazar Can we close this or is there some additional data needed?

MarceloSalazar, 2018-08-21:
Hi @AnttiKauppila , there is only one minor item pending: https://github.com/ARMmbed/mbed-os-5-docs/issues/551#issuecomment-395737183 regarding the LoRaPHY vs LoRaRadio.

I don't think we understand the difference and should expand/clarify this. It looks to me both are very low-level drivers.
Also not sure how they map to the Mbed OS architecture (from user API, network stacks, to low-level drivers), for example in mesh we have: Socket API, Nanostack, 15.4 MAC driver.

Would you be able to help? Thanks!

AnttiKauppila, 2018-08-21:
@MarceloSalazar Would something like this be more clear in docs:
---
LoRaPhy is handling LoRaRadio parameters which are same regardless of radio being used, like for example regional parameters. LoRaRadio should be just controlling the registers and interrupts. This way we can keep actual LoRaRadio drivers rather small.
---

What comes to 15.4 radio the use case is almost exactly the same. Application creates 15.4 driver and then passes it to Nanostack. There are some common logic which is being done and the driver is just controlling the registers and interrupts.

If you agree with above text we can update the documentation. Or you can suggest something better. Or if this is still not clear please try to open up the difficult parts.

MarceloSalazar, 2018-08-21:
Thanks @AnttiKauppila , this clarification is what I was looking for :)
Can you please send a PR to enhance the docs, then we can go ahead and close this ticket. Thanks!

SeppoTakalo, 2018-08-21:
Hmm..
I was confused as well because Nanostack has *Phy which is the driver. But this has a *Radio which is the actual driver and it is then controlled by the *Phy.

But comparison is pointless, this is a different API. So as long as it is clear after seeing the example and porting guide, then its fine.

AnotherButler, 2018-08-27:
Closing because PR merged; please feel free to reopen if you feel this does not satisfy the requirements

closed 2018-08-27 by AnotherButler