ARMmbed/mbed-os-5-docs #553: Mbed statistics examples shown in document

MarceloSalazar, 2018-06-01:
Reviewed Mbed statistics section in PR (in the meantime it shows up in docs)
https://github.com/deepikabhavnani/Handbook/blob/2324689b7a5f20fa494995178e3fd787b5f86b57/docs/reference/api/platform/MbedStats.md

The document is looking good. Just a question on examples shown in the document.

> Memory statistics
> Thread statistics
> System information
> CPU statistics

The examples listed at the bottom seem to be Mbed 2 and hardcoded to specific versions:

https://os.mbed.com/teams/mbed_example/code/mbed-os-example-platform-utils/file/92b97ba04fd3/main.cpp
https://os.mbed.com/teams/mbed_example/code/thread_statistics/file/8cfc3eff0d78/main.cpp/
https://os.mbed.com/teams/mbed_example/code/system_information/file/8189a62cbb4e/main.cpp/
https://os.mbed.com/teams/mbed_example/code/cpu_stats/file/de30f5166672/main.cpp/
https://os.mbed.com/teams/mbed_example/code/cpu_usage_calculation/file/358f7cf7196d/main.cpp/

Maybe a question for @adbridge 

Can we point at the corresponding example here:
https://os.mbed.com/teams/mbed-os-examples/code
(mirrored automatically from github)

adbridge, 2018-06-01:
@MarceloSalazar As previously discussed we do not update examples in the teams directory. This is the responsibility of Zach Shelby's team, please see email chain I just forwarded to you.

MarceloSalazar, 2018-06-01:
@adbridge as per @ashok-rao comments today:

> Examples look trivial. Code snippets would be better there?

This would be great as accelerates the integration of this document with the official Handbook, and reduces the maintenance effort for the future.

@AnotherButler what do you think?

AnotherButler, 2018-06-01:
We don't like hardcoded code snippets in our documentation. They're difficult to update, and they lack the functionality of code in teams. We want users to be able to click the **Import to Compiler** button and be able to import any of our examples into the Arm Mbed Online Compiler. They can't do that with hardcoded snippets.

adbridge, 2018-06-01:
@AnotherButler  I disagree with this. For very small snippets of code that are pointless as examples then being an embedded part of the documentation is the right solution AFAIAC. Keeping the example snippet in the code up to date is no different to keeping the actual doc itself up to date....

AnotherButler, 2018-06-01:
I'm fine with hardcoding a few lines of code, but three of these examples have 60. That's a full example, and I'd like to keep our full example functionality.

As a temporary solution for 5.9, we can hardcode these, but if we do that, we need to discuss how we'll handle examples in the future.

MarceloSalazar, 2018-06-01:
I think we can easily have max 3 or 4 lines of code to show the usage of the corresponding APIs.

@deepikabhavnani is this something you can do?

deepikabhavnani, 2018-06-01:
@MarceloSalazar @adbridge 
Examples are already available in git repo, as per @ashok-rao that was needed for OOB testing as well.
We need to mirror them at https://os.mbed.com/teams/mbed-os-examples/

deepikabhavnani, 2018-06-01:
> I think we can easily have max 3 or 4 lines of code to show the usage of the corresponding APIs.

Sorry @MarceloSalazar we are constantly creating examples and replacing them with existing snippets. So this will not be acceptable solution. 

None of my examples are Mbed OS 2, all of them link to last release of Mbed OS 5, which is limitation of online tools hence kept private.

I will remove old examples from "mbed-os-examples", if GIT examples are mirrored, if not done on time then I can update existing examples once release is created and then update them to be public

adbridge, 2018-06-01:
@deepikabhavnani You need to ask @screamerbg to add your examples to his mirroring script.

adbridge, 2018-06-05:
@SenRamakri Have we reached a conclusion about what we need to do here for the release?

SenRamakri, 2018-06-06:
@adbridge - The examples are now available in mbed-os-examples.

AnotherButler, 2018-06-06:
Is it OK to close this issue?

adbridge, 2018-06-07:
@MarceloSalazar Are you happy to close this now ?

closed 2018-06-07 by MarceloSalazar