ARMmbed/mbed-os-5-docs #559: Bootloader tutorial should explain application header

chuanga, 2018-06-03:
Review bootloader.md page (https://github.com/ARMmbed/Handbook/blob/5.8/docs/tutorials/bootloader.md) as part of mbed-os-5.9). Cannot access master branch; so look at 5.8 branch instead.

Document should explain fields of application header or have a link to relevant document if available.

AnotherButler, 2018-06-04:
You should be reviewing 5.9 documents on our staging site, which requires being on the network. Please see this page: https://os-doc-builder.test.mbed.com/docs/development/tutorials/bootloader.html

AnotherButler, 2018-06-04:
To raise a PR against the correct branch, please use the new_engine branch. This goes to our staging site. We do not work on master.

AnotherButler, 2018-06-06:
We exposed the link to relevant documentation. Is it OK to close this issue?

MarceloSalazar, 2018-06-08:
Closing. @chuanga please reopen if you see an issue.

closed 2018-06-08 by MarceloSalazar