ARMmbed/mbed-os-5-docs #566: Bootloader tutorial would benefit from more specific step by step instructions

maclobdell, 2018-06-04:
The Tutorial at https://github.com/ARMmbed/Handbook/blob/new_engine/docs/tutorials/bootloader.md, doesn't flow like a true tutorial.  It is more like an overview of each important topic.  If someone is actually trying to walk through this document, they will need to understand many steps that are missing.  

Most of the information is available at the README in this repo:  https://github.com/ARMmbed/mbed-os-example-bootloader

I recommend adding information from the *Building the bootloader* and *Building the application with the bootloader* sections.  

At a minimum, this documentation page should reference the example bootloader repos we have available.  

generic mbed os bootloader example:
https://github.com/ARMmbed/mbed-os-example-bootloader

mbed cloud bootloader
https://github.com/ARMmbed/mbed-bootloader

maclobdell, 2018-06-05:
After reviewing this with Russ, I think it is fine to just highlight more prominently the link to the bootloader application.

c1728p9, 2018-06-05:
@maclobdell I created #575 to make the link more prominent and remove old content.

adbridge, 2018-06-07:
@maclobdell Please confirm you are happy with the changes and close this if so.

maclobdell, 2018-06-08:
LGTM.  Thanks!

closed 2018-06-08 by maclobdell