ARMmbed/mbed-os-5-docs #581: Update link to Mbed CLI Windows installer

MarceloSalazar, 2018-06-08:
Doc: https://github.com/ARMmbed/Handbook/blob/new_engine/docs/tutorials/quickstart/quick-start-cli.md

--> Download and run the [Mbed CLI Windows .exe installer] 

Should have a new link to *mbed-media.mbed.com ... (TBC) * and indicate which version is available (latest right now is 0.4.7 corresponding to Mbed CLI 1.7.1)

Also a reminder on pending task to open this repo: https://github.com/ARMmbed/mbed-cli-windows-installer
So we don't have to update this doc every time there is a new release of Mbed CLI Windows installer. See IOTDOCS-410

@arekzaluski @AnotherButler

AnotherButler, 2018-06-08:
@arekzaluski What's the most recent link I should use?

MarceloSalazar, 2018-06-13:
@arekzaluski  can you please comment?

arekzaluski, 2018-06-13:
@AnotherButler @MarceloSalazar @ashok-rao The latest link to installer is available here:
https://mbed-media.mbed.com/filer_public/00/c2/00c2bc59-21c5-4ea2-9916-f9c38e5b2aec/mbed_installer_v047.exe

It should be updated only if it finishes successfully OOB testing. @MarceloSalazar @ashok-rao any news about it?

MarceloSalazar, 2018-06-13:
OOB with latest CLI installer has finished successfully - please update the link.

MarceloSalazar, 2018-06-18:
@AnotherButler thanks for the fix.

I found out this page still points to an old Mbed CLI installer.
https://os.mbed.com/docs/v5.9/tools/installing-with-the-windows-installer.html

Can we please update the link asap? People out there will need the latest link. Thanks!

AnotherButler, 2018-06-18:
This change has now gone live. Please feel free to reopen if you feel it has not been fully resolved.

closed 2018-06-18 by AnotherButler