ARMmbed/mbed-os-5-docs #582: [OoB_5.9]: Error handling docs needs few clarifications.

ashok-rao, 2018-06-08:
Overall very good information in the docs. Good job @SenRamakri 

However, a number of clarifications needed as part of this issue. @AnotherButler or @SenRamakri could you please check?

From the handbook: https://os-doc-builder.test.mbed.com/docs/development/reference/error-handling.html

1 . the line: "extract information from error status values and to generate fatal system errors that terminate the application."
It is not clear to me what "..generate fatal system errors" refers to. Are we saying error codes and macros can be used to "generate" the error i.e. deliberately introduce the error - what for? or is it intended to say "..and to **RE-**generate fatal system errors.." RE-as in reproduce errors that will be useful for debugging the actual crash? Could this be made clear?

2. "Mbed OS system error status values can also capture the module reporting the error by encoding the module identifier in the error status, depending on the type of error status." -> Can you please add some examples here?

3. " In the case of warnings, it is recorded by the system, and you can retrieve it later.." -> from where? how? Can you please add a few lines for this?

4. "Note that file name capture is disabled by default." -> Where to enable this? Might be worth adding a "TIP" here as to where can this be enabled.

5. "We provide APIs to retrieve errors or warnings from the Error history and the first and last errors." -> Where are these? Can we link them here, please?

SenRamakri, 2018-06-08:
PR created with fixes - https://github.com/ARMmbed/Handbook/pull/586

AnotherButler, 2018-06-11:
Closing this as resolved because #586 merged. Please feel free to reopen if you feel this is not resolved.

closed 2018-06-11 by AnotherButler