ARMmbed/mbed-os-5-docs #597: Migrating 2 => 5 is missing Online Compiler instructions

janjongboom, 2018-06-19:
https://os.mbed.com/docs/v5.8/tutorials/migrating.html should have a section for the Online Compiler too. It's even a lot easier on the Online Compiler.

janjongboom, 2018-06-19:
F.e. this could be the instructions (but it needs a screenshot):

In the Online Compiler you do:

1. Right click on `mbed`, select *Delete...*.
1. Right click on your project, select *Import Library > From URL*.
1. Enter `https://github.com/armmbed/mbed-os`.

And done, now the application is updated to latest version of Mbed OS. Hope this helps.

HarishkumarBringiramakannu, 2018-06-19:
Hi @janjongboom ,
Thankyou for your link. It was very helpful. But now I am struck with this error "Error: Library name 'platform' is not unique (defined in '/tmp/chroots/ch-2619ff6b-a6f9-4624-81ff-1ac9c3c14ea0/extras/mbed/TARGET_DISCO_L072CZ_LRWAN1/mbed_lib.json' and '/tmp/chroots/ch-2619ff6b-a6f9-4624-81ff-1ac9c3c14ea0/src/mbed-os/platform/mbed_lib.json')"

when I was searching in internet, many got this error but they don't know the solution, do you know how to clear this ? could you help me out through this ?

AnotherButler, 2018-06-27:
@elm3 Is there any way we could get content help with this from @yennster (who wrote the original document) or someone else from your team?

yennster, 2018-06-27:
@AnotherButler @janjongboom Fixed here: https://github.com/ARMmbed/mbed-os-5-docs/pull/609

closed 2018-07-09 by janjongboom