ARMmbed/mbed-os-5-docs #598: Question: specification for build output file name

jflynn129, 2018-06-19:
Is there a macro or some method of naming the output file name that is generated when a build occurs from mbed compile?  For example, if I my project is in a directory named "test" but I want the executable to be named "my_first_test" is there a way to specify the name via the --output in the linker step? currently it looks like it always takes the name of the current project directory and uses it.

it looks like make.py creates a bin_file string that is used but it isn't obvious if there is a parameter you can place in the mbed_app.json file that will allow you to simply configure the output file name.

jflynn129, 2018-06-19:
sorry, again after digging/googling I was able to find the answer to this question.  it doesn't appear to be in the official documentation though...

closed 2018-06-19 by jflynn129

AnotherButler, 2018-06-19:
@theotherjimmy Could you please help me add this to the official documentation?

theotherjimmy, 2018-06-19:
@AnotherButler It is in the official documentation. Maybe we need another link to it. It's the `artifact_name` section of `mbed_app.json`.
https://github.com/ARMmbed/mbed-os-5-docs/blob/development/docs/reference/configuration/configuration.md#mbed_appjson-specification

jflynn129, 2018-06-19:
IMO, artifact_name doesn't have the same connotation or definition as "executable_name" so it took me a while (and a lot of googling) to figure out that "executable_name" was "artifact_name" in mbed'speak. there is also no obvious example of its use--you should add it into the code spinet below that table.

it sure would be helpful to have a glossary or document that listed all the optional/possible keywords that are available for a user to modify/configure via the mbed_app.json and mbed_lib.json files...

theotherjimmy, 2018-06-19:
I can add it to the example.

@jflynn129 I think I just liked exactly this:
> optional/possible keywords that are available for a user to modify/configure via the mbed_app.json and mbed_lib.json files...

Looking upward, what you're asking for is a combination of what I linked and `mbed compile --config ...`

theotherjimmy, 2018-06-19:
https://github.com/ARMmbed/mbed-os-5-docs/blob/development/docs/reference/configuration/configuration.md#examining-available-configuration-parameters

theotherjimmy, 2018-06-20:
@jflynn129 Where else would you expect to find this information regarding renaming your executable?

jflynn129, 2018-06-21:
Where it is at is fine but I would provide more explanation that a sentence fragment in a table. I would also add the elements of the table to the code sample below the table to reinforce the explanation.

PR #602 (2018-06-21): Add `artifact_name` section to mbed_app example
Resolves #598