ARMmbed/mbed-os-5-docs #601: Add description of "detect_code" field

deepikabhavnani, 2018-06-20:
detect_code field (part of targets.json) description is missing at https://docs.mbed.com/docs/mbed-os-api/en/latest/api/md_docs_mbed_targets.html

AnotherButler, 2018-06-27:
@SenRamakri This looks like a code fix. Could someone from your team please help with this?

deepikabhavnani, 2018-07-05:
@AnotherButler - This does not look like code issue.

Document should have brief description of elements described in below section
 
```
2  "inherits": ["Target"],
3  "core": "Cortex-M4",
4  "extra_labels": ["Freescale", "K20XX", "K20DX256"],
5  "OUTPUT_EXT": "hex",
6  "is_disk_virtual": true,
7  "supported_toolchains": ["GCC_ARM", "ARM"],
8  "post_binary_hook": {
9  "function": "TEENSY3_1Code.binary_hook",
10  "toolchains": ["ARM_STD", "ARM_MICRO", "GCC_ARM"]
11  },
12  "progen": {"target": "teensy-31"},
13  "detect_code": ["0230"]
```

Some elements from this section are skipped like `OUTPUT_EXT` `detect_code` and `is_disk_virtual`

kegilbert, 2018-08-03:
@AnotherButler @deepikabhavnani Can we close this? As Jimmy pointed out the linked doc is out of date, completely missed the URL sorry about that.

https://os.mbed.com/docs/v5.9/tools/adding-and-configuring-targets.html should have the missing info.

deepikabhavnani, 2018-08-03:
@kegilbert - New document covers OUTPUT_EXT , but nothing about 'detect_code and is_disk_virtual'

We should have atleast a statement explaining what those config options are for.

'''
Modules and Boards may have the following keys:
supported_form_factors.
is_disk_virtual.
detect_code.
extra_labels.
public.
config.
overrides.
forced_reset_timeout.
''''

aashishc1988, 2018-09-27:
PR https://github.com/ARMmbed/mbed-os-5-docs/pull/751 filed

AnotherButler, 2018-10-04:
Because PR #751 has merged, I'm closing this issue. Please feel free to reopen if it has not been resolved to your satisfaction.

closed 2018-10-04 by AnotherButler