ARMmbed/mbed-os-5-docs #604: mbed-os documentation: expand all sections + download in pdf

0xc0170, 2018-06-21:
### Description

@AnotherButler Moved from https://github.com/ARMmbed/mbed-os/issues/7236

The mbed-os documentation here https://os.mbed.com/docs/latest/introduction/index.html is very nice but it is not easy to read or find information. It would be very useful to have these 2 features:

- be able to expand all the sections in order to see all titles at once
- be able to download the entire documentation in a pdf file

### Issue request type

[ ] Question  
[X] Enhancement  
[ ] Bug  

cc @bcostm

AnotherButler, 2018-06-21:
@mariannecrowder @iriark01 Could y'all please add this feedback to the other UX feedback you've collected?

cederom, 2018-06-24:
+1 here :-) I would also like / prefer that way of Handbooking :-)

iriark01, 2018-06-25:
This is a duplicate of work already on the web team's backlog.

IMPORTANT NOTICE: The contents of this email and any attachments are confidential and may also be privileged. If you are not the intended recipient, please notify the sender immediately and do not disclose the contents to any other person, use it for any purpose, or store or copy the information in any medium. Thank you.

iriark01, 2018-06-27:
Closing as duplicate to existing work (already planned)

closed 2018-06-27 by iriark01