ARMmbed/mbed-os-5-docs #607: documentation version switch + "current" documentation

cederom, 2018-06-24:
Hello world :-)

Sometimes I am finding a reference to something in MbedOS from Google or external site.. then it could show some older version of documentation.. when switching the version to a current one the URL is rewritten always to the documentation main page instead just rewrite the version number of the currently watched site..

For instance when I read about BLE in `v5.6` and want to switch to `v5.9` I still would like to stay on that BLE page for version `v5.9`.

Maybe a "current version" (`current`) documentation would solve a bit the problem of outdated links.. as the documentation would always show the most up to date "current" version of documentatnion, while user could easily select older version if necessary..

AnotherButler, 2018-06-25:
I will pass this request to our web team.

Please note that because of content changes and reorganizations, there is not always a 1:1 mapping of content from one version to another.

You can replace "v5.9" with "latest" in your URL to stay on the most recent version of published documentation.

cederom, 2018-06-25:
Thank you @AnotherButler :-)

"latest" works fine and I think it would be good solution to prevent outdated links :-)

Very nice big banner is showed on top of very old documentation that is also referenced on the web :-)

The problem is still switching between versions that redirects to a main documentation page instead currently read article but for another version :-)

Have a nice day and send greetings to The Team :-)

AnotherButler, 2018-07-10:
@cederom I've forwarded your feedback. Also, the teams says hi. Everyone hopes you're doing well :)

closed 2018-07-10 by AnotherButler