ARMmbed/mbed-os-5-docs #611: Configure debugger with VSCode

masoudr, 2018-06-28:
Hi,
I'm trying to configure debugger using Openocd 0.10.0 with VSCode from [this reference](https://os.mbed.com/docs/v5.8/tutorials/visual-studio-code.html) on Windows 10. I'm using `STLink` with an `STM32F407` custom board. I can run `Openocd` and `arm-none-eabi-gdb` manually with the exact configuration without any problem but on VS Code it seems that there are problems in configuration and throw an exception on after dubug starts on any code like `mbed-os-example-blinky`. Here are configuration and error output.
`launch.json`

```
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "C++ Launch",
            "type": "cppdbg",
            "request": "launch",
            "program": "${workspaceRoot}/BUILD/ARCH_MAX/GCC_ARM/${workspaceRootFolderName}.elf",
            "args": [],
            "stopAtEntry": true,
            "cwd": "${workspaceRoot}",
            "environment": [],
            "externalConsole": false,
            "debugServerArgs": "-f C:/openocd-0.10.0/openocd-0.10.0/scripts/interface/stlink-v2.cfg -f C:/openocd-0.10.0/openocd-0.10.0/scripts/target/stm32f4x_stlink.cfg -c init -c \"reset init\"",
            "serverLaunchTimeout": 20000,
            "filterStderr": true,
            "filterStdout": false,
            "serverStarted": "target halted due to debug-request, current mode: Thread",
            "preLaunchTask": "mbed",
            "setupCommands": [
                { "text": "-target-select remote localhost:3333", "description": "connect to target", "ignoreFailures": false },
                { "text": "-file-exec-and-symbols ${workspaceRoot}/BUILD/ARCH_MAX/GCC_ARM/${workspaceRootFolderName}.elf", "description": "load file", "ignoreFailures": false},
                { "text": "-interpreter-exec console \"monitor endian little\"", "ignoreFailures": false },
                { "text": "-interpreter-exec console \"monitor reset\"", "ignoreFailures": false },
                { "text": "-interpreter-exec console \"monitor halt\"", "ignoreFailures": false },
                { "text": "-interpreter-exec console \"monitor arm semihosting enable\"", "ignoreFailures": false },
                { "text": "-target-download", "description": "flash target", "ignoreFailures": false }
            ],
            "logging": {
                "moduleLoad": true,
                "trace": true,
                "engineLogging": true,
                "programOutput": true,
                "exceptions": true
            },
            "linux": {
                "MIMode": "gdb",
                "MIDebuggerPath": "/usr/bin/arm-none-eabi-gdb",
                "debugServerPath": "pyocd-gdbserver"
            },
            "osx": {
                "MIMode": "gdb",
                "MIDebuggerPath": "/usr/local/bin/arm-none-eabi-gdb",
                "debugServerPath": "pyocd-gdbserver"
            },
            "windows": {
                "preLaunchTask": "mbed",
                "MIMode": "gdb",
                "MIDebuggerPath": "C:\\Program Files (x86)\\GNU Tools ARM Embedded\\4.9 2015q3\\bin\\arm-none-eabi-gdb.exe",
                "debugServerPath": "C:\\openocd-0.10.0\\openocd-0.10.0\\bin-x64\\openocd.exe",
                "setupCommands": [
                    { "text": "-environment-cd ${workspaceRoot}\\BUILD\\ARCH_MAX\\GCC_ARM" },
                    { "text": "-target-select remote localhost:3333", "description": "connect to target", "ignoreFailures": false },
                    { "text": "-file-exec-and-symbols ${workspaceRootFolderName}.elf", "description": "load file", "ignoreFailures": false},
                    { "text": "-interpreter-exec console \"monitor endian little\"", "ignoreFailures": false },
                    { "text": "-interpreter-exec console \"monitor reset\"", "ignoreFailures": false },
                    { "text": "-interpreter-exec console \"monitor halt\"", "ignoreFailures": false },
                    { "text": "-interpreter-exec console \"monitor arm semihosting enable\"", "ignoreFailures": false },
                    { "text": "-target-download", "description": "flash target", "ignoreFailures": false }
                ]
            }
        }
    ]
}
```
`tasks.json`
```
{
    // See https://go.microsoft.com/fwlink/?LinkId=733558
    // for the documentation about the tasks.json format
    "version": "0.1.0",
    "name": "mbed",
    "isShellCommand": true,
    "showOutput": "always",
    "problemMatcher": {
        "owner": "cpp",
        "fileLocation": ["relative", "${workspaceRoot}/mbed-os"],
        "pattern": {
            "regexp": "^(.*):(\\d+):(\\d+):\\s+(warning|error):\\s+(.*)$",
            "file": 1,
            "line": 2,
            "column": 3,
            "severity": 4,
            "message": 5
        }
    },
    "args": ["compile", "--profile=debug", "-t", "GCC_ARM", "-m", "ARCH_MAX"],
    "linux": {
        "command": "mbed"
    },
    "osx": {
        "command": "mbed"
    },
    "windows": {
        "command": "mbed"
    }
}
```

Error(on main block):
> C continue: {"threadId":-2}
> 
> 1: (220259) <--exec-continue
> 
> 1: (220266) ->^running
> 
> 1: (220266) ->*running,thread-id="all"
> 
> 1: (220266) ->(gdb)
> 
> 1: (220266) ->&"\n"
> 
> 1: (220266) ->^done
> 
> 1: (220266) ->(gdb)
> 
> openocd: Info : halted: PC: 0x08008cbc
> 1: (220952) ->=breakpoint-modified,bkpt={number="1",type="breakpoint",disp="keep",enabled="y",addr="0x08008cba",func="main()",file=".\\source\\main.cpp",fullname="C:\\Users\\Masoud\\Desktop\\mbed_test\\mbed-os-example-blinky\\source\\main.cpp",line="8",thread-groups=["i1"],times="2",original-location="main"}
> 
> 1: (220954) ->~"\n"
> 
> 
> 1: (220954) ->~"Breakpoint 1, main () at .\\source\\main.cpp:8\n"
> 
> Breakpoint 1, main () at .\source\main.cpp:8
> 1: (220955) ->~"8\t        led1 = !led1;\n"
> 
> 8	        led1 = !led1;
> 1: (220955) ->*stopped,reason="breakpoint-hit",disp="keep",bkptno="1",frame={addr="0x08008cba",func="main",args=[],file=".\\source\\main.cpp",fullname="C:\\Users\\Masoud\\Desktop\\mbed_test\\mbed-os-example-blinky\\source\\main.cpp",line="8"},thread-id="1",stopped-threads="all"
> 
> 1: (220955) <-1024-stack-list-frames 0 1000
> 
> 1: (220965) ->1024^done,stack=[frame={level="0",addr="0x08008cba",func="main",file=".\\source\\main.cpp",fullname="C:\\Users\\Masoud\\Desktop\\mbed_test\\mbed-os-example-blinky\\source\\main.cpp",line="8"}]
> 
> 1: (220965) ->(gdb)
> 
> 1: (220965) 1024: elapsed time 10
> 
> 1: (220966) ->&"\n"
> 
> 1: (220966) ->^done
> 
> 1: (220966) ->(gdb)
> 
> C threads: null
> 
> C stackTrace: {"threadId":-2,"startFrame":0,"levels":20}
> 
> 1: (220976) <-1025-stack-list-arguments 0 0 0
> 
> 1: (220987) ->1025^done,stack-args=[frame={level="0",args=[]}]
> 
> 1: (220987) ->(gdb)
> 
> 1: (220987) 1025: elapsed time 10
> 
> 1: (220987) ->&"\n"
> 
> 1: (220987) ->^done
> 
> 1: (220987) ->(gdb)
> 
> C scopes: {"frameId":1000}
> 
> 1: (221014) <-1026-stack-list-variables 0
> 
> 1: (221020) ->1026^done,variables=[]
> 
> 1: (221020) ->(gdb)
> 
> 1: (221020) 1026: elapsed time 5
> 
> 1: (221020) ->&"\n"
> 
> 1: (221020) ->^done
> 
> 1: (221020) ->(gdb)
> 
>

AnotherButler, 2018-06-28:
@elm3 Is this something that someone from your team could help with?

masoudr, 2018-06-28:
The error seems that it has some breakpoints I tried to put a breakpoint before the exception line and somehow it didn't throw the exception. I'm not sure if it is a bug or something with VS code.

yennster, 2018-07-02:
@masoudr Do you still have an issue?

masoudr, 2018-07-02:
@yennster yes I have. I just put a breakpoint in the last line of code but obviously this is not a fix. I've tested current tools (I mean both openocd an gdb) on other frameworks like `platformio` with the exact configuration without any problem. I think there is something wrong with the `vs code` configuration or the elf file generated by current mbed-os.

yennster, 2018-07-03:
@masoudr Okay, thanks for the clarification. I'll try to reproduce the issue on my end. 👍

yennster, 2018-07-16:
@masoudr I am unable to reproduce the issue. Can you provide more information on how you are exporting to VSCode? Have you made your own custom target within the Mbed OS library? Are you using the online compiler or the Mbed CLI?

masoudr, 2018-07-16:
I'm using the exact steps for import the mbed-blinky project with `mbed-cli` then I just added the mbed-os 5 support for `seeed-arch-max` board (It has the exact chip `STM32F407VET6`) in `targets.json` file by adding "5" to `release_versions`:
```
    "ARCH_MAX": {
        "inherits": ["FAMILY_STM32"],
        "supported_form_factors": ["ARDUINO"],
        "core": "Cortex-M4F",
        "supported_toolchains": ["ARM", "uARM", "GCC_ARM"],
        "program_cycle_s": 2,
        "extra_labels_add": ["STM32F4", "STM32F407", "STM32F407xG", "STM32F407VG"],
        "device_has_add": ["ANALOGOUT"],
        "release_versions": ["2", "5"],
        "device_name": "STM32F407VG"
    },

```
Afterwards I export the project:
`mbed export -i vscode_gcc_arm -m ARCH_MAX --profile mbed-os/tools/profiles/debug.json`
After that I open folder by vscode and modify `launch.json` and `tasks.json` as I wrote on the first post.
I'm using the latest version of `openocd` from [this link](http://gnutoolchains.com/arm-eabi/openocd/).
I tried to use the exact steps on `Ubuntu 16.04` and the same problem happened to me.
I tried to change my target to `DISCO_F407VG` with no luck.
I tried other versions of `GNU ARM Compiler`.
The strange thing in the error is that I can see the `breakpoint` on line 8 but there is no breakpoints in the code:

> 1: (4982) ->(gdb)
> 1: (5383) ->=breakpoint-modified,bkpt={number="1",type="breakpoint",disp="keep",enabled="y",addr="0x08008ca6",func="main()",file=".\\source\\main.cpp",fullname="C:\\Users\\Masoud\\Desktop\\mbed_test\\mbed-os-example-blinky\\source\\main.cpp",line="8",thread-groups=["i1"],times="1",original-location="main"}
> 1: (5384) ->~"\nBreakpoint "
> 1: (5384) ->~"1, main () at .\\source\\main.cpp:8\n"
> 1: (5385) ->~"8\t        led1 = !led1;\n"
> 1: (5385) ->*stopped,reason="breakpoint-hit",disp="keep",bkptno="1",frame={addr="0x08008ca6",func="main",args=[],file=".\\source\\main.cpp",fullname="C:\\Users\\Masoud\\Desktop\\mbed_test\\mbed-os-example-blinky\\source\\main.cpp",line="8"},thread-id="1",stopped-threads="all"
> Breakpoint 1, main () at .\source\main.cpp:8
> 8	        led1 = !led1;
> 1: (5389) <-1021-stack-list-frames 0 1000
> 1: (5395) ->1021^done,stack=[frame={level="0",addr="0x08008ca6",func="main",file=".\\source\\main.cpp",fullname="C:\\Users\\Masoud\\Desktop\\mbed_test\\mbed-os-example-blinky\\source\\main.cpp",line="8"}]
> 1: (5395) ->(gdb)
> 1: (5395) 1021: elapsed time 6
> 1: (5395) ->&"\n"
> 1: (5395) ->^done
> 1: (5395) ->(gdb)

`main.cpp`:
```
#include "mbed.h"

DigitalOut led1(PA_6);

// main() runs in its own thread in the OS
int main() {
    while (true) {
        led1 = !led1;
        wait(0.5);
    }
}
```

yennster, 2018-07-26:
@masoudr Are you using a USB hub to connect your board to the computer? And if you are is it a powered USB hub or passive?

masoudr, 2018-07-27:
@yennester No, my board uses a separate power supply and the ST-Link connected directly to my PC.

AnotherButler, 2018-08-27:
Has this issue been resolved?

yennster, 2018-08-28:
@masoudr I apologize for the delay on this issue. I'm taking another look at it today.

AnotherButler, 2018-09-28:
I'm going to close this issue because it seems to be code related and not docs related. Please feel free to reopen if this is not the case.

closed 2018-09-28 by AnotherButler

camelator, 2018-11-06:
I have the same issue as well

AnotherButler, 2018-11-06:
@TacoGrandeTX Is your team tracking this issue internally? I don't want it to get lost, but I don't think this is a good place to track it.

TacoGrandeTX, 2018-11-06:
Yes -  @yennster created a tracking JIRA (IOTDEV-1558).  I've just updated it with additional notes... thanks for the notification!