ARMmbed/mbed-os-5-docs #620: [Request] Document explaining Mbed OS priorities

yennster, 2018-07-03:
I'm not sure if this already exists, but I am looking for a doc/tutorial that explains how Mbed OS sets priority to Interrupts, Threads, etc. So far I can only find documentation regarding these topics on the forums, so I think having an official doc would be useful. Here's some community examples of information I would like to see in this doc (I'm not sure of the accuracy of the following documents as they are really old, but these are just examples of the ideas I'd like to see discussed):

- https://os.mbed.com/forum/mbed/topic/1913/
- https://os.mbed.com/forum/mbed/topic/26506/
- https://os.mbed.com/forum/mbed/topic/3732/
- https://os.mbed.com/users/AjK/notebook/regarding-interrupts-use-and-blocking/
- https://os.mbed.com/users/simon/code/Priority/docs/2b952ccae54d/main_8cpp_source.html

I know Thread has an Mbed OS API call that allows you to set priority levels: https://os.mbed.com/docs/v5.9/reference/thread.html

However, I understand that you can also call the CMSIS function `NVIC_SetPriority` to set your interrupts priority level directly, so some explanation on that would be nice as well.

AnotherButler, 2018-07-10:
In our published documentation, we have: https://os.mbed.com/docs/v5.9/tutorials/application-flow-control.html
And: https://os.mbed.com/docs/v5.9/tutorials/the-eventqueue-api.html#prioritization

We also have this document, which is currently being updated, so it's not live: https://github.com/ARMmbed/mbed-os-5-docs/blob/development/docs/reference/api/thread_safety.md

Are any of those what you're looking for, or should we add a document like this to our list of content to create?

AnotherButler, 2018-08-27:
ping @yennster

closed 2018-08-28 by yennster