ARMmbed/mbed-os-5-docs #638: Documentation doesn't state whether FATFileSystem is thread-safe

AnotherButler, 2018-07-24:
As requested on [the forum]((https://os.mbed.com/forum/mbed/topic/32521/)), please add the following content to the FATFileSystem user documentation:

When using FATFileSystem on an SD card (via sd-driver), is it thread-safe?

Answer: Yes, the FATFileSystem is thread safe. However, the synchronization is actually implemented in the C++ layer: https://github.com/ARMmbed/mbed-os/blob/master/features/filesystem/fat/FATFileSystem.cpp#L681-L690

ChanN's FF_FS_REENTRANT option would allow us to synchronize the filesystem through the ChaN C API, but synchronizing in the C++ layer allows us to directly use the C++ RTOS APIs and use the same locks for the C++ class's state.

bmcdonnell-ionx, 2018-07-25:
I was the OP of the [forum question](https://os.mbed.com/forum/mbed/topic/32521/).

If `FATFileSystem`'s thread safety is media-independent, please document as such. Maybe don't even mention `sd-driver`/`SDBlockDevice`.

theamirocohen, 2018-08-21:
Updatetd handbook documentation in PR #674

AnotherButler, 2018-08-27:
@bmcdonnell-ionx Does this resolve your issue?

closed 2018-08-27 by AnotherButler