ARMmbed/mbed-os-5-docs #673: MBRBlockDevice::partition - type

deepikabhavnani, 2018-08-21:
https://github.com/ARMmbed/mbed-os/issues/5506

Query:
I am looking documentation https://os.mbed.com/docs/v5.6/mbed-os-api-doxy/class_m_b_r_block_device.html#ace6dd73e8514a0a2afab8f8bc37bada8

I tried to find supported values/filesystems for type, which is "8-bit partition type to identitfy partition's contents". Would you clarity usage of parameter type with MBRBlockDevice::partition? What are supported values? Is it type same than Partition ID in table https://en.wikipedia.org/wiki/Partition_type#List_of_partition_IDs?

Response:

Yes, the type parameter is an ID from that list:
https://en.wikipedia.org/wiki/Partition_type#List_of_partition_IDs

I didn't include it initially because it's such an unwieldy list (>200 options), but if it helps I can add an enum of the most common types.

I'm also thinking it would be a good idea to just use 0x83 as a default.

Docs Update:
I propose that we clarify mbed-os documentation about usage of partition type.

AnotherButler, 2018-12-06:
@deepikabhavnani Have we updated this in our documentation?

deepikabhavnani, 2018-12-06:
@AnotherButler - No

closed 2020-06-05 by 0xc0170