ARMmbed/mbed-os-5-docs #692: Release schedule missing

0xc0170, 2018-08-29:
https://github.com/ARMmbed/mbed-os-5-docs/blob/576bd6262b25c695848fec1abfd21571e23a8a1a/docs/introduction/release_process.md 

This should contain information when we release (patch release, minor release and also mbed 2 release)

cc @ARMmbed/mbed-os-maintainers @ashok-rao

AnotherButler, 2018-08-29:
@0xc0170 The release cadence is included for patch, feature and major releases: https://github.com/ARMmbed/mbed-os-5-docs/blob/development/docs/introduction/release_process.md

Information about Mbed 2 releases should go in the Mbed 2 documentation.

0xc0170, 2018-09-03:
> @0xc0170 The release cadence is included for patch, feature and major releases: https://github.com/ARMmbed/mbed-os-5-docs/blob/development/docs/introduction/release_process.md

There's only "every two weeks". I should be added "every odd week on Wednesday" for  the patch release. The minor -  I dont see there how often it is released - should be added "every quarter" or similar.

> Information about Mbed 2 releases should go in the Mbed 2 documentation.

Where is it ? I would like to add the cadence there as well

AnotherButler, 2018-09-11:
The feature release section states "Feature releases occur once a quarter."

The Mbed 2 docs are here: https://os.mbed.com/handbook/Homepage

0xc0170, 2018-09-17:
> The feature release section states "Feature releases occur once a quarter."

👍 

> The Mbed 2 docs are here: https://os.mbed.com/handbook/Homepage

@adbridge Can you update the homepage to define the mbed 2 release cadence there?

adbridge, 2018-09-17:
@0xc0170 Done.

AnotherButler, 2018-09-18:
@0xc0170 Is this ready to close?

0xc0170, 2018-09-19:
Yes, now it is.

closed 2018-09-19 by 0xc0170