ARMmbed/mbed-os-5-docs #707: Pelion Device Management Client required for mbed dm functionality

dhwalters423, 2018-09-05:
The current documentation for the `mbed dm` command makes no mention of Pelion Device Management Client (PDMC). 

https://github.com/ARMmbed/mbed-os-5-docs/blob/development/docs/tools/offline/cli-device-management.md

In order for someone to use the `mbed dm` command to update a device,their device must be running an application with PDMC and a supported managed or unmanaged bootloader.

Documentation needs to mention that this command only works for devices running Pelion Device Management Client and the application is combined with a bootloader. Additionally, you may want to have a line similar to:

"To test this feature, you can run the [Pelion Device Management Client Example](https://github.com/ARMmbed/mbed-cloud-client-example) on one of our reference platforms."

dhwalters423, 2018-09-06:
One more suggestion related to this, we should mention that the device needs to be running the `<name of app>_update.bin` first. So in the tutorial we should say

1. run `mbed dm init` command
2. `mbed compile ...`
3. Flash device with `<name_of_app_update.bin>`
5. View serial terminal for device id
4.  `mbed dm update device -D <device ID> -m <target>`
6.  View serial terminal for update progress

ashok-rao, 2018-09-11:
@kegilbert / @AnotherButler : Could you please address this? Thanks.

kegilbert, 2018-09-11:
I'm double checking with some cloud/pelion people, I can update the docs accordingly.

oliverjharper, 2018-09-14:
@teetak01 - any chance you could help with @AnotherButler question above?

AnotherButler, 2018-09-14:
@teetak01 Do you agree with @dhwalters423 that we need to specify which binary one uses to flash the device first that contains the bootloader, and which binary one uses for an update? If so, could you please help me clarify that?

teetak01, 2018-09-17:
The "default" binary should be flashed to the board. Then the "_update.bin" is automatically used by mbed-cli for firmware update, but if customer wants to provide the bin explicitly, it should be the _update.bin.

closed 2018-09-20 by AnotherButler